neuoy/OneWireArduinoSlave #2: Respond to RESET

aivs, 2015-10-20:
Ill trying to use you library to emulate DS18B20.
To start communication with One Wire master I need to respond to Reset. 
I see Reset in console, then I see Error. But data from master I don't see.
How to respond to RESET?

void owReceive(OneWireSlave::ReceiveEvent evt, byte data) {
	switch (evt) {
    Serial.println(data, HEX);
    // Detect Reset on bus
    case OneWireSlave::RE_Reset:
        Serial.println("Reset");
    case OneWireSlave::RE_Byte:
      // Read Scratchpad [BEh] command
      if (data == 0xBE) { 
      	 Serial.println("Scratchpad command");
      }
    case OneWireSlave::RE_Error:
         Serial.println("Error");
		OneWire.write(&acknowledge, 1, NULL);
		break;
	default:
		; 
	}
}

neuoy, 2015-10-20:
Hi,

Have you tried without writing "Reset" and data to the serial port? Writing to the serial port is really slow compared to some 1-wire events that occur within a few micro-seconds. More precisely, it should be fine to write to the serial port from your main loop, but don't do it from the owReceive handler (or any function called from it).

In case that doesn't solve your issue, can you tell me what arduino board and 1-wire master are you using? So far I've only tested with an Arduino Uno and a DS9490R.

aivs, 2015-10-22:
For test I use another Arduino with OneWire library - http://www.pjrc.com/teensy/td_libs_OneWire.html
and example to read DS18B20 http://cxem.net/arduino/arduino17.php.

Without example it's very difficult to understand this is my problem or library problem.

Now in RE_Byte I only assign data to a command variable, and then in loop I'm bringing data to Serial, it always 0
```
    case OneWireSlave::RE_Byte:
      // Read Scratchpad [BEh] command
      command = data;
```

neuoy, 2015-10-23:
OK, maybe the library is not compatible with the other arduino library (things can be sensitive to different timings). Just in case, are you sure the electronic part is correct ? (I think you need a resistor between the bus and 5V that will keep the bus at 5V unless either the master or the slave pulls it low)

I'll try to make a DS18B20 implementation, that would surely be a good example to help understand how to use the library, and also allow easier testing in other configurations (different arduino board, or different master). But I can't say when that'll be, so it will probably not solve your issue on short term.

aivs, 2015-10-23:
I use resistor 4.7kohm between data and +5V. And when I use real DS18B20 with this schematics all works fine.
I will try to use oscilloscope.
About volatile variable its a good tip, my var was not volatile.
Thanks for quick answers!

I'm making the emulator ds18b20 just for fun, so
If this works, capabilities of my z-wave sensor will be expanded and it cool!

aivs, 2015-10-24:
Hey!! it works, Im stupid, DS18B20 Rom should be start with byte 0x10.
Now I have good communication between 2 arduino.
Ready sketch I will post later.
Library works fine!

neuoy, 2015-10-24:
Great, that's reassuring me for compatibility with other masters :-)

closed 2015-10-24 by neuoy