neuoy/OneWireArduinoSlave #3: Answer with HIGH

aivs, 2015-10-29:
From DS18B20 Datasheet:
If the DS18B20 is powered by an external supply, the master can issue “read time slots” (see the
1-Wire Bus System section) after the Convert T command and the DS18B20 will respond by transmitting
0 while the temperature conversion is in progress and 1 when the conversion is done.

Please show me right way how on 1-wire line set HIGH level with you library?
digitalWrite(2,HIGH) - is it right way?

neuoy, 2015-10-29:
There is currently no publicly exposed function to send or receive a single bit, but that would indeed be an interesting functionality. There are internal functions to send/receive bits, that are used by bytes send/receive functions, so it's mostly an interface problem, the functionality is already implemented. I'll take a look to see how these functionalities can be exposed.

I wouldn't recommend to directly modify the pin output, because properly sending a bit is a little more involved (you have to wait for the master signal, then respond with the bit), but maybe that would work anyway. However, you never have to write a HIGH level, because the wire is at 5V by default (thanks to the pullup resistor). The simple fact of doing nothing is enough to send a 1 bit. In the case you want to write a 0 bit however, the bus must be pulled low, but you have to wait for the master signal to do it, otherwise you don't know when to release the bus, and if you never release the bus there won't be any further communication. Also, trying to directly write on the bus will necessarily interfere with the library that thinks it has exclusive control of it. The library would actually think it's the master misbehaving.

aivs, 2015-10-29:
I think you get a lot of experience with the implementation of DS18B20 ))

neuoy, 2015-12-15:
I've added functions in the library to send individual bits (with the possibility to repeat the same bit until new order), and also to be notified of individual bits received (but that's not of use in the case of a DS18B20 implementation). To test everything, I've modified the sample sketch to implement a DS18B20, this should demonstrate how to use most features of the library with a concrete example.

closed 2015-12-15 by neuoy

aivs, 2015-12-16:
Cool!!