neuoy/OneWireArduinoSlave #9: Same code works in Uno and not in Mega2560

elusive-code, 2016-12-05:
Hello, I have something rather wierd.

I'm trying to join several arduinos in 1-wire network.
I wrote simple test bellow to check stuff - it just checks device presense.
It contains both code for both master (BOARD_NUMBER=1) and slave(everything else). 

Problem is, when I use Arduino Uno as a slave - it works fine. 
However when I upload same code to Arduino Mega 2560 and use it in the same setup master node doesn't see slave (slave initialized though - printed it's address in serial).

```
#include <OneWire.h>
#include <OneWireSlave.h>

#define BOARD_NUMBER 4
#define DEVICE_TYPE 0
#define PIN 3
#define MAX_DEVICES 10

byte address[8] = {};

void generate_address(){
    randomSeed(analogRead(A0));
    address[0] = DEVICE_TYPE;
    for (int i=0; i<7; i++){
        address[i] = random();
    }
    address[7] = OneWire::crc8( address, 7);     
}

#if BOARD_NUMBER==1
    //master
    
    OneWire ds(PIN);
    byte slaveAddress[MAX_DEVICES][8] = {0};
    
    void setup_onewire(){
        
    }

    void update_onewire(){
        
        ds.reset();
        ds.reset_search();

        int found = 1;
        int i;

        for( i=0; i<MAX_DEVICES && found; i++) {
            found = ds.search(slaveAddress[i]);            
        }
        
        if (i<=1) {
             Serial.println("No one on the line");
             ds.reset_search();
             delay(1000);
             return;
        }
        unsigned long pollStart = millis();
        
        digitalWrite(LED_BUILTIN, HIGH);

        for (int j=0; j<i-1; j++) {
            Serial.print("Found device ");
            for( int k = 0; k < 8; k++) {
                Serial.print(slaveAddress[j][k], HEX);
                Serial.print(" ");
            }
            
            uint8_t crc = ds.crc8(slaveAddress[j],7);
            if ( crc==slaveAddress[j][7] ){
                Serial.print("\tmatch");
            } else {
                Serial.print("\tno match!");
            }
            Serial.println();
            
            ds.reset();
            ds.select(slaveAddress[j]);
            ds.write(BOARD_NUMBER,0);

            int bufLen = 6;
            byte buf[bufLen];
            ds.read_bytes(buf,bufLen);
            int data = buf[0];
            /*
            int data = ds.read();
            int ff1 = ds.read();
            int ff2 = ds.read();
            byte checksum[2]={ds.read(),ds.read()};
            */
            
            Serial.print("Sent ");
            Serial.print(BOARD_NUMBER);
            Serial.print(" received ");
            for( int k = 0; k < bufLen; k++) {
                Serial.print(buf[k], HEX);
                Serial.print(" ");
            }
            //Serial.print(data);

            bool check = ds.check_crc16(buf,3,&buf[3]);
            Serial.print(" checksum match: ");
            Serial.println(check);
            
            
        }
        digitalWrite(LED_BUILTIN, LOW);
        unsigned long pollEnd = millis();
        Serial.print("Polled ");
        Serial.print(i-1);
        Serial.print(" devices in ");
        Serial.print(pollEnd-pollStart);
        Serial.println(" ms");
        delay(1000);
    }
#else
    //slave
    byte buf[1] = {BOARD_NUMBER};
    
    void receive(OneWireSlave::ReceiveEvent evt, byte data) {
        if (evt!=OneWireSlave::RE_Byte) return;
        OWSlave.write((const byte*)buf,1);
    }

    void setup_onewire(){
        //OWSlave.setReceiveCallback(&receive);
        OWSlave.begin(address, PIN);        
    }

    void update_onewire(){
        delay(10);
    }
    
#endif

void setup() {    
    Serial.begin(9600);
    generate_address();
    setup_onewire();
    Serial.print("One wire address: ");
    for( int i = 0; i < 8; i++) {
        Serial.print(address[i], HEX);
        Serial.print(" ");
    }
    Serial.println();
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);  
}

void loop() {
    update_onewire();
}
```

neuoy, 2017-04-30:
If someone makes it work on other boards that Arduino Uno (the only board I've tested), I'm interested to know ; but otherwise I can't help myself since I don't have the hardware to test, nor the motivation to work on it.

I leave the issue open so others can know which boards support are most requested.

alexandrewillame, 2018-01-11:
I would also be very interested for a support for the ATMega2560 board!