neuoy/OneWireArduinoSlave #10: Add support for STM32??

timpur, 2017-04-24:
It would be nice to see this lib support more boards :)

Some one has altered you code to make it work for STM32 (http://www.stm32duino.com/viewtopic.php?f=9&t=1852)

STM32 using a timer class which makes it tricky to add support for this lib.
but i was thinking if you upgraded the whole lib to use a timer class for the timer functions than you could add support for STM32.

if you used this lib (http://playground.arduino.cc/Code/Timer1) then you could make it easy to port feature boards too, since there will be a common interface.

The STM32 (https://github.com/rogerclarkmelbourne/Arduino_STM32) uses mostly the same doc as this http://docs.leaflabs.com/docs.leaflabs.com/index.html.

neuoy, 2017-04-24:
This seems interesting, I already thought about using a timer library, just didn't get the time to do it yet. This could be a good occasion.

timpur, 2017-04-29:
I tried doing it my self, but still dont have my head wrapped around MCU's and timers and this low level hardware stuff so i dont quite understand how to replace you existing code with the timer library stuff.

If you found the time to make this upgrade, id at least really appreciate it, but also this world make porting this library to any new device easy :).

neuoy, 2017-05-03:
I've made a test with the Timer1 library, but although it seems to work as expected when I make simple timer tests, once integrated in the 1-wire code it doesn't work. For example the device search algorithm fails. I'm not in the mood of grabbing my oscilloscope to debug what happens right now, so I'll postpone that idea. If you're interested I've pushed a branch "timer1" with the non-working code (compiles, run, but 1-wire communications do not work). I'm not sure if I'll make anything more on this subject in the future; I'm quite happy with the timer details being centralized in LowLevel.h (which is a recent change, I just merged a port of the library to attiny85 from another user's fork), though I must admit having a timer library that is already ported to multiple platforms would be interesting.

timpur, 2017-05-05:
Thats a bummer, but if i get around to it ill continue on with your work. Yeah it would be nice :) 

Thanks for all your efforts.