neuoy/OneWireArduinoSlave #12: Use sample code with ds2482-800

ferec, 2017-07-06:
Hello guys,

I am trying to figure out how to use this library with ds2482. I have a working code for communicating with a real DS18b20 thermometer, it does not work with this lib. In the detection phase I've got invalid commands on Arduino side like 0x65 or 0xd9. Maybe some timing problem?

I am using Arduino Uno with original board. DS2482 communicates with a short cable directly, Raspberry Pi B+ on the I2C side. This communication works with different 1-wire ICs like DS2450, DS2408.

Thank you for your suggestions.

Ferenc

neuoy, 2017-07-08:
Hi Ferenc,

Sorry for the late reply, and thanks for the update. I'll check the 1-wire specs and the other masters reset timing, and see if I can find a condition that satisfies all the masters tested with the library at this time (i.e. another arduino as master, which depends on the library used on the other arduino, or a DS9490R, or a DS2482 in your case). I suppose it's possible since standard 1-wire slaves seem to work fine in all these situations.

ferec, 2017-07-09:
Hello,

It looks problem is not in library itself, just in the sample code for ds18b20.

neuoy, 2017-07-09:
Hmm, do you mean that, while performing the conversion, the fake DS18B20 pulls the bus low, which resets itself? That would mean it doesn't work with other masters either? I'll take a closer look at what it's supposed to do when I have some time.

But that would not explain why you got invalid commands, like 0x65 or 0xd9?

ferec, 2017-07-09:
At first time I added some Serial code into sample causing random bits/bytes received. After reinstalling the library and trying the sample code it still did not work. Then I used oscilloscope and logical analyzer to figure out whats going on. Then I realized that the fake pull-down triggers the reset event on bus.