neuoy/OneWireArduinoSlave #13: Read ROM [0x33]	case 0x33: // READ ROM

austintxjim, 2017-07-12:
I can not get the Read ROM [0x33] command to work in the example.  The Search ROM [0xF0], Match ROM [0x55], and Skip ROM [0xCC] are working with the master I am using.

I am new to Arduino but have some experience with C and I can not determine where the problem is.

I am using an Arduino Uno Rev 2. board (which should behave same as Rev 3 in this case).  Can you give me a clue?   Thanks!!   Great example code, easy to read.

austintxjim, 2017-07-12:
I guess I should give a few more details, when I set the ROM ID to:
owROM[7] = { 0x28, 0x55, 0x55, 0x55, 0x55, 0x55, 0x02 };

I get { 0x08, 0x55, 0x11, 0x54, 0x01, 0x00, 0x00, 0xAF} using the Read ROM command [0x33], but I get the expected string back with the Search ROM [0xF0] command.

neuoy, 2017-07-13:
Hi,

To be honest, I don't remember having tested the read rom command, maybe there's a bug. Can you otherwise transmit bytes in both directions, is it only this command that doesn't work? What master are you using?

austintxjim, 2017-07-13:
Hi,

The other commands work (I haven't tested all of them, I mostly need to read and write using 'skip ROM', and that works).   I can transmit bytes in both directions.   For now I am testing the software with a DS9097U-E25.  http://datasheets.maximintegrated.com/en/ds/DS9097U-009-DS9097U-S09.pdf

I am using that for debugging purposes before I put the Arduino in my project.

Thanks!!

austintxjim, 2017-07-13:
I guess there is one other 'quirk', which doesn't affect me, I mean if you want 'perfection', and all that.

The software always gives up it's address (ID) using the 'Search ROM' [0xF0] regardless of the device family.   I believe it should only respond if it sees '0x28' as a family code from the controller.   This is a non issue for me, It was just something I observed while testing.

neuoy, 2017-07-13:
Thanks for these reports. I'm not sure there is a problem with the search algorithm, it will respond only if the master selected all the bits corresponding to its ROM so far (it will drop from the search after the first bit that diverges). But I haven't looked at it for a long time, so maybe I've missed or don't remember about something. For the read ROM command however this looks like a bug, if you say you have no problem otherwise to transmit bytes, it means it's not a timing issue with your master, so it's probably a bug in the library. Not sure if/when I'll find time to look at it though.

austintxjim, 2017-07-13:
I am using some 'canned' software for  the search algorithm.   With an iButton the function returns an error if the wrong family code is used, But I never get an error with the Arduino, It returns the correct SN with the family code and checksum included.   I agree that if the Arduino 'drops the search' after the first incorrect bit, then my 'canned software' should return an error or at least incorrect bytes.  I don't know how to explain it (yet).  I'll take out a logic analyzer and look at the bits and give an update, probably tomorrow.

Thanks for taking the time to write this great utility!!!   Very interesting!  It gives me many ideas.

austintxjim, 2017-07-18:
Got it working!  Whew.   It was a timing problem (in my case) with the reset pulse.

I made the following change to line 14 of the .cpp code:

const unsigned long ResetMinDuration = 420; //  was 480;

It works perfectly now.

closed 2017-07-18 by austintxjim